LimHyungTae/ERASOR #6: catkin_erasor/src/ERASOR/src/offline_map_updater/main_kitti.cpp:8:67: error: use of deleted function ‘erasor::OfflineMapUpdater::OfflineMapUpdater(const erasor::OfflineMapUpdater&)’      erasor::OfflineMapUpdater updater = erasor::OfflineMapUpdater();

GXU-doudou, 2021-11-29:
Hello, I encountered this error when compiling the program, but I don't know how to solve it. 
If you can help, I will be grateful.

LimHyungTae, 2021-11-29:
Hello, could you give me whole compling messages?

GXU-doudou, 2021-11-30:
@LimHyungTae Thanks for your reply~The specific information of the error is as follows：
Errors     << erasor:make /home/doudou/catkin_erasor/logs/erasor/build.make.000.log                                                                                                                        
/home/doudou/catkin_erasor/src/ERASOR/src/offline_map_updater/main_kitti.cpp: In function ‘int main(int, char**)’:
/home/doudou/catkin_erasor/src/ERASOR/src/offline_map_updater/main_kitti.cpp:8:67: error: use of deleted function ‘erasor::OfflineMapUpdater::OfflineMapUpdater(const erasor::OfflineMapUpdater&)’
     erasor::OfflineMapUpdater updater = erasor::OfflineMapUpdater();
                                                                   ^
In file included from /home/doudou/catkin_erasor/src/ERASOR/src/offline_map_updater/main_kitti.cpp:2:0:
/home/doudou/catkin_erasor/src/ERASOR/include/erasor/OfflineMapUpdater.h:9:11: note: ‘erasor::OfflineMapUpdater::OfflineMapUpdater(const erasor::OfflineMapUpdater&)’ is implicitly deleted because the default definition would be ill-formed:
     class OfflineMapUpdater {
           ^
/home/doudou/catkin_erasor/src/ERASOR/include/erasor/OfflineMapUpdater.h:9:11: error: use of deleted function ‘std::unique_ptr<_Tp, _Dp>::unique_ptr(const std::unique_ptr<_Tp, _Dp>&) [with _Tp = ERASOR; _Dp = std::default_delete<ERASOR>]’
In file included from /usr/include/c++/5/bits/locale_conv.h:41:0,
                 from /usr/include/c++/5/locale:43,
                 from /usr/include/c++/5/iomanip:43,
                 from /usr/include/boost/math/policies/error_handling.hpp:12,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/melodic/include/ros/time.h:58,
                 from /opt/ros/melodic/include/ros/ros.h:38,
                 from /home/doudou/catkin_erasor/src/ERASOR/src/offline_map_updater/main_kitti.cpp:1:
/usr/include/c++/5/bits/unique_ptr.h:359:7: note: declared here
       unique_ptr(const unique_ptr&) = delete;
       ^
c++: internal compiler error: Killed (program cc1plus)

LimHyungTae, 2021-11-30:
I guess it's about the version issue. Could you give me some information of your machine? Such as Ubuntu version, ros version, boost version, and so forth.

GXU-doudou, 2021-11-30:
I think so, version info:
Ubuntu 18.04.6 LTS
rosversion melodic 1.14.11 
libboost1.65-dev:amd64: /usr/include/boost/version.hpp
gcc version 7.3.0 (GCC) 
python3.7

GXU-doudou, 2021-12-03:
I have solved the problem, thanks for your help. @LimHyungTae

closed 2021-12-03 by GXU-doudou

zxczrx123, 2022-02-17:
> I have solved the problem, thanks for your help. @LimHyungTae

Hi, I had the same problem. Can you share how you solved it?

GXU-doudou, 2022-02-17:
这是来自QQ邮箱的假期自动回复邮件。您好，邮件已经收到，我会及时查看，辛苦你了。

fentuoli, 2022-07-28:
> > I have solved the problem, thanks for your help. @LimHyungTae
> 
> Hi, I had the same problem. Can you share how you solved it?

Hi, I also had the same problem :), have you solved this problem?