LimHyungTae/ERASOR #7: Assertion `cloud.points.size () == cloud.width * cloud.height' failed.

zxl19, 2022-01-13:
Hi @LimHyungTae ! Thank you for your excellent work. I ran the demo on my computer and the result looks great. However, when I tried ERASOR on my own dataset, I encountered the following error:

```shell
main_in_your_env: /usr/include/pcl-1.8/pcl/conversions.h:252: void pcl::toPCLPointCloud2(const pcl::PointCloud<PointT>&, pcl::PCLPointCloud2&) [with PointT = pcl::PointXYZI]: Assertion `cloud.points.size () == cloud.width * cloud.height' failed.
```

This happens on a specific frame. I checked the corresponding `.pcd` file but it looked alright. I also tried this [suggestion](https://answers.ros.org/question/45170/what-does-assertion-cloudpointssize-cloudwidth-cloudheight-failed-mean/) but it did not work. Could you please give me some hints on how to solve this, thanks!

LimHyungTae, 2022-01-16:
Hello there, it's because I used `emplace_back()` haha...Originally, `pc.push_back(pt)` automatically manage its width and height, yet if you use `pc.points.emplace_back()`, it doesn't work.

Could you add some lines as follow?:

```
map_to_be_saved.width  = map_to_be_saved.points.size();
map_to_be_saved.height = 1;
```

Thank you for your kind attention! :)

zxl19, 2022-01-16:
Hi @LimHyungTae ! Thank you for your reply. I checked the code and found those lines in `OfflineMapUpdater::callback_flag()`, it seems that I have already merged pull request #2. The error still occurs when ERASOR processed to a specific frame, I am still looking for the cause :( Also, does ERASOR require the `.pcd` files have ring channels?

zxczrx123, 2022-02-21:
@zxl19 Hi, have you solved the problem and successfully run it on your own data?

LimHyungTae, 2022-03-01:
@zxl19 Hello there, did you solve your problem? Your experiences would be a great help for me!!!

May-sysu, 2022-03-20:
hi，i have the same problem in my data. have you solved it ? i found in the erasor_utils :: cloud2msg code, cloud.size() isnt equal to cloud.heigth plus cloud.width in some specific frame

zxiaomzxm, 2022-03-29:
Hi, there, same error here. 
And as @LimHyungTae  and @May-sysu said, I add
```
cloud.width  = cloud.points.size();
cloud.height = 1;
```
in ` erasor_utils :: cloud2msg` function before `pcl::toROSMsg(cloud, cloud_ROS);`
and get things done.^ ^

May-sysu, 2022-03-29:
> Hi, there, same error here. And as @LimHyungTae and @May-sysu said, I add
> 
> ```
> cloud.width  = cloud.points.size();
> cloud.height = 1;
> ```
> 
> in ` erasor_utils :: cloud2msg` function before `pcl::toROSMsg(cloud, cloud_ROS);` and get things done.^ ^

May-sysu, 2022-03-29:
@zxiaomzxm yep, we can do that to solve this error. however, i find this cases doesnt exit if you change smaller Lmax . i dont know why? maybe that is why happened in some specific frame

zxl19, 2022-03-29:
Hi everyone @LimHyungTae @zxczrx123, sorry for the late reply. I've been busy with my thesis and not really checked my GitHub account these days. I still didn't find the actual cause, but it looks like during processing ERASOR returned an empty point cloud. So I manually set the width and height to zero in `/include/tools/erasor_utils.hpp` and the problem didn't appear anymore.

```cpp
template<typename T>
sensor_msgs::PointCloud2 cloud2msg(pcl::PointCloud<T> cloud, std::string frame_id = "map")
{
    std::cout << "cloud.points.size() = " << cloud.points.size() << std::endl;
    std::cout << "cloud.height() = " << cloud.height << std::endl;
    std::cout << "cloud.width() = " << cloud.width << std::endl;
    cloud.height = 0;
    cloud.width = 0;
    sensor_msgs::PointCloud2 cloud_ROS;
    pcl::toROSMsg(cloud, cloud_ROS);
    cloud_ROS.header.frame_id = frame_id;
    return cloud_ROS;
}
```

Best Wishes.

closed 2022-04-15 by zxl19