LimHyungTae/ERASOR #8: catkin build error: can not find erasor/node.h.

zxczrx123, 2022-02-18:
Thanks for your great work!  According to your introduction, I encountered the following problem. I can't find node.h from the source file. @LimHyungTae
[image]
Could you please give me some help? 🥺

LimHyungTae, 2022-03-01:
Hmm....well could you build the repo after comment out of all `add_executable`, `target_link_libraries`, and `add_dependencies `lines? Then, msg would be successfully compiled. Then, please recompile this repo again!

zxczrx123, 2022-03-14:
Thanks for your reply! Actually, I have solved this problem by using `catkin_make`.

closed 2022-03-14 by zxczrx123