LimHyungTae/ERASOR #9: how to get poses_lidar

cml3028, 2022-02-28:
hello,how to get poses_lidar when I run run_erasor_in_your_env.launch?

LimHyungTae, 2022-03-01:
Hello, I'll share some additional github repo called lio4erasor!

cml3028, 2022-03-01:
> Hello, I'll share some additional github repo called lio4erasor!

Thank you for your reply！I will try it.

cml3028, 2022-03-01:
> Hello, I'll share some additional github repo called lio4erasor!

sorry,I can not find any github repo named lio4erasor. Is it correct?

LimHyungTae, 2022-03-01:
Oops, sorry for bothering you... I mean "I will"....please wait a few days!!!

cml3028, 2022-03-01:
> Oops, sorry for bothering you... I mean "I will"....please wait a few days!!!

OK，I see. Thank you!

chery-5, 2023-12-19:
I also want to know how  to get poses_lidar. I want to remove the pedestrian indoors, but when I input the pcds and poses from FAST-LIO2，there is no effect . It seems that the point cloud is not aligned with the global map. But I have no idea about it.

jingyilon, 2024-07-19:
> I also want to know how to get poses_lidar. I want to remove the pedestrian indoors, but when I input the pcds and poses from FAST-LIO2，there is no effect . It seems that the point cloud is not aligned with the global map. But I have no idea about it.

hello！Has this issue been resolved？