LimHyungTae/ERASOR #11: use HDL 32E Velodyne's  data

pytorch1100, 2022-03-04:
Hi @LimHyungTae， thank you for your great work! I have one question. If i use Velodyne 32E's data, what parameters should be changed? The parameters of Lidar are below. 
[image]

pytorch1100, 2022-03-04:
The distance between lidar and the ground is shown in the figure below, which is 1.8m. If I want to change the min_h in the code, can I use  min_h = -1.8m?
[image]

closed 2022-03-14 by pytorch1100

darshanAtSwaayatt, 2023-11-28:
Hi @pytorch1100,

Were you able to figure out what parameters to change for HDL 32E Velodyne lidar?