LimHyungTae/ERASOR #15: How to deal with dynamic point clouds without ground ?

LeisureLei, 2022-06-22:
Hi @LimHyungTae , thanks for your sharing work. Your methods can remove dynamic point clouds in normal ground-scene efficicetly. But there are not always ground in lidar frame especially for submap which is build incremently. How to deal with this case?

LimHyungTae, 2022-06-22:
Hi, I didn't fully understand the meaning of "especially for submap which is built incrementally". Could you give me some more details?

LeisureLei, 2022-06-23:
> Hi, I didn't fully understand the meaning of "especially for submap which is built incrementally". Could you give me some more details?

That means building submap incrementally while detecting dynamic of submap. When a new scan coming, I use scan and submap to detect dynamic and insert scan to submap next.

LimHyungTae, 2022-11-11:
Sorry for the late reply, I think ERAOSR does not support that situation 'cause ERASOR strongly assumes that ground points are observable.