LimHyungTae/ERASOR #16: When I run "roslaunch erasor run_erasor.launch target_seq:="05" ",it has no response

fentuoli, 2022-07-04:
[image]

Then the interface stays still. Could you give me some advice about how to solve this probles?

LimHyungTae, 2022-11-11:
Hello, I'm about to do defense on 16th, Nov. So, after that day, I'll check it!

fentuoli, 2022-11-21:
> Hello, I'm about to do defense on 16th, Nov. So, after that day, I'll check it!

I have solved this problem, thanks!

Lonely52Hz, 2023-02-13:
> > Hello, I'm about to do defense on 16th, Nov. So, after that day, I'll check it!
> 
> I have solved this problem, thanks!

Hi, is it possible for me to know how you solve it? I met the same problem. Thanks.

Sanguan94, 2024-04-24:
怎么解决的  谢谢

Sanguan94, 2024-04-24:
> > Hello, I'm about to do defense on 16th, Nov. So, after that day, I'll check it!
> 
> I have solved this problem, thanks!

bro how to solve this problem?

Sanguan94, 2024-04-24:
> > > Hello, I'm about to do defense on 16th, Nov. So, after that day, I'll check it!
> > 
> > 
> > I have solved this problem, thanks!
> 
> Hi, is it possible for me to know how you solve it? I met the same problem. Thanks.

Have you solved the problem?