LimHyungTae/ERASOR #20: Eigen::internal::variable_if_dynamic<T, Value>::variable_if_dynamic(T) [with T = long int; int Value = 3]: Assertion `v == T(Value)' failed.

xiaoxiaolu1111, 2022-09-06:
main_in_your_env_ros: /usr/include/eigen3/Eigen/src/Core/util/XprHelper.h:110: Eigen::internal::variable_if_dynamic<T, Value>::variable_if_dynamic(T) [with T = long int; int Value = 3]: Assertion `v == T(Value)' failed.
[main_in_your_env_ros-1] process has died [pid 32655, exit code -6, cmd /home/dsj/catkin_erasor/devel/lib/erasor/main_in_your_env_ros __name:=main_in_your_env_ros __log:=/home/dsj/.ros/log/4f1bf28e-2dac-11ed-a109-d05099fb0c81/main_in_your_env_ros-1.log].
log file: /home/dsj/.ros/log/4f1bf28e-2dac-11ed-a109-d05099fb0c81/main_in_your_env_ros-1*.log

LimHyungTae, 2022-11-11:
Hello, I'm about to do defense on 16th, Nov. So, after that day, I'll check it!

LimHyungTae, 2022-11-11:
What command do you run?

zhumenga, 2022-11-16:
I changed  this line ： tf4x4_cam.topLeftCorner<3, 3>(0, 0) = q.toRotationMatrix(); in file main_in_your_env.cpp  to 
  tf4x4_cam.topLeftCorner(3, 3) = q.normalized().toRotationMatrix();  then  it works.

MIke-1118, 2022-11-17:
> I changed this line ： tf4x4_cam.topLeftCorner<3, 3>(0, 0) = q.toRotationMatrix(); in file main_in_your_env.cpp to tf4x4_cam.topLeftCorner(3, 3) = q.normalized().toRotationMatrix(); then it works.

I followed your method and tried. It happens again with the following errors. Have you encountered it?

[ INFO] [1668568723.312462663]:  PASS! 
[ INFO] [1668568723.411885393]: 1303th frame is comming
[ INFO] [1668568723.490956884]: 1014759=1014759| 138259 + 876500
[ INFO] [1668568723.607537384]:  Extracting VoI takes 0.1828s
main_in_your_env_ros: /usr/include/pcl-1.8/pcl/conversions.h:252: void pcl::toPCLPointCloud2(const pcl::PointCloud<PointT>&, pcl::PCLPointCloud2&) [with PointT = pcl::PointXYZI]: Assertion `cloud.points.size () == cloud.width * cloud.height' failed.

zhumenga, 2022-11-17:
Add two lines： cloud.width  = cloud.points.size();  cloud.height = 1;   to function cloud2msg  in file erasor_utils.hpp :