LimHyungTae/ERASOR #24: Remove much more static points(false positive) than expectation

fentuoli, 2022-11-22:
Hello, I'm sorry to bother you. I test your method in my own datasets, but I found the false positive points is very large. In the raw map, the stat. pts is 844762, and in the estimated map, the est.stat.pts is 475379, which leading the preservation rate is very low. Is it because of the parameter setting that causes this situation? How to deal with it?
     I'm waiting for your reply!
Best!

heroacool, 2023-05-26:
I have encountered the same problem as well. ERASOR almost removed all the point clouds on the tree canopy.
before filter
[image]

after filter
[image]

Kin-Zhang, 2024-07-22:
I also observed this issue when benchmarking all methods: https://github.com/KTH-RPL/DynamicMap_Benchmark and reported ERASOR in the paper. Feel free to check. I think it's common.

There are some conservative methods (mainly the ray-casting method) you can try like: https://github.com/Kin-Zhang/octomap, https://github.com/KTH-RPL/dufomap

LimHyungTae, 2024-08-12:
Thanks @Kin-Zhang, for the kind follow-up! Since our primary goal was to make a static map for a) localization and b) path planning/following purposes (and we also checked that kinds of loss of static points do not affect navigation quality).

LimHyungTae, 2024-08-12:
Even I am convinced that removing the crowns of trees sometimes helps long-term localization!

Kin-Zhang, 2024-08-12:
> Even I am convinced that removing the crowns of trees sometimes helps long-term localization!

Thanks for sharing your opinion. 

I think people are talking about <static points, too much false positive> in a more **general way**, and crowns of trees are one example. You can imagine others if you think crowns are not important, but think about if too much static structure is removed. Sometimes it is fine, sometimes not. It depends a lot on your data and afterward, map usage.

Like the example in DUFOMap, we have two-floor data, I think you can imagine how EARSOR performs already as it's highly height threshold-based. Feel free to run [the data](https://github.com/KTH-RPL/DynamicMap_Benchmark?tab=readme-ov-file#dataset--scripts).
[image]

gzyabc, 2024-12-23:
Hi, I would like to ask you about the way you used the erasor algorithm on your dataset.
The way to use rosbag or the way to generate pcd?
I used the latter, and there was a bit of a mismatch between the cloud frames and the map when using erasor.
I want to know how you handled it. Thank you for your reply.

LimHyungTae, 2024-12-25:
Hello, then what about
1) interpolation using timestamps
2) and then run a static map building algorithm?

Thanks

gzyabc, 2024-12-25:
Thank you for your reply. Maybe I didn't put it right. I use KITTI data sets, and use your Shared saveEachPosesAndDeskewedPcs method for point cloud. In this case, a misalignment occurred. You don't have this problem with the 16 rows of liosam LiDAR data you provided. What's the problem?
I will record my own data set in the future, but my current idea is to use KITTI's data set to go over the process of culling my own data set. Unfortunately, there is a misalignment. Please advise me. @LimHyungTae

LimHyungTae, 2026-05-22:
Following up on this thread for visitors who land here later —

I just landed two PRs that touch this directly:

- **PR #38** retunes the KITTI configs against the paper-GT and adds a small NumPy-2-safe PR/RR evaluator (`scripts/analysis_runner.py`) that also reports the `est → GT` nearest-neighbor distance distribution. Worth checking that overlap first: if less than ~98% of your est points sit within half a voxel of the GT, the input map or pose convention is misaligned and PR/RR will be misleading.
- **PR #39** expands the "In your own dataset" section with a coordinate-frame caution block. The most common cause of "too many false positives on my own data" I've seen is a pose-convention mismatch (the CSV is `T_map_from_lidar`, not `T_lidar_from_map` — see the new README block).

If your `est → GT` overlap is fine and PR is still low, the main tuning knob is `scan_ratio_threshold`: lower it (try `{0.10, 0.13, 0.15, 0.20}`) and watch RR — for Seq 00 on KITTI, going from `0.20 → 0.10` recently moved F1 from 0.948 to 0.957 with negligible RR cost.

@gzyabc — for the misalignment you saw, the most likely cause is that `saveEachPosesAndDeskewedPcs` writes poses in a different convention than `pose · pcds/<i>.pcd → map`. Could you check whether overlaying `pose_0 · pcds/000000.pcd` on `dense_global_map.pcd` lines up visually? If not, the pose convention is the issue — see the new README block.