LimHyungTae/ERASOR #31: roslaunch erasor run_erasor.launch target_seq:="05"

DoongLi, 2023-09-11:
When I run roslaunch erasor run_erasor.launch target_seq:="05", I encountered the following error, and my system is ubuntu20.04

roslaunch erasor run_erasor.launch target_seq:="05"
... logging to /home/doongli/.ros/log/195c78c0-504d-11ee-ba99-4f48c8864c97/roslaunch-doongli-Precision-5820-Tower-X-Series-990898.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/__init__.py", line 347, in main
    p.start()
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/parent.py", line 305, in start
    self._start_infrastructure()
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/parent.py", line 254, in _start_infrastructure
    self._load_config()
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/parent.py", line 156, in _load_config
    self.config = roslaunch.config.load_config_default(self.roslaunch_files, self.port,
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/config.py", line 461, in load_config_default
    loader.load(f, config, argv=args, verbose=verbose)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 763, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 735, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 665, in _recurse_load
    if ifunless_test(self, tag, context):
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 74, in ifunless_test
    if_val, unless_val = obj.opt_attrs(tag, context, ['if', 'unless'])
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 208, in opt_attrs
    return [self.resolve_args(tag_value(tag,a), context) for a in attrs]
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 208, in <listcomp>
    return [self.resolve_args(tag_value(tag,a), context) for a in attrs]
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/xmlloader.py", line 189, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/substitution_args.py", line 378, in resolve_args
    return _eval(arg_str[7:-1], context)
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/substitution_args.py", line 347, in _eval
    return str(eval(s, {}, _DictWrapper(context['arg'], functions)))
  File "<string>", line 1
    arg('target_seq') ==01
                         ^
SyntaxError: leading zeros in decimal integer literals are not permitted; use an 0o prefix for octal integers

niuwengang, 2023-11-06:
just use x instead of  0x, for example as following
```
<group if="$(eval arg('target_seq') ==5)">
    <rosparam command="load" file="$(find erasor)/config/seq_05.yaml" />
</group>
<group if="$(eval arg('target_seq') ==7)">
    <rosparam command="load" file="$(find erasor)/config/seq_07.yaml" />
</group>
```

fwailiu, 2024-04-24:
@DoongLi  Have you had any success running your own dataset with erasor?、

DoongLi, 2024-04-24:
Nope

Sanguan94, 2024-04-24:
[When I run "roslaunch erasor run_erasor.launch target_seq:="05" ",it has no response]    do you know how to  solve this problem?thanks

closed 2024-05-06 by DoongLi