LimHyungTae/ERASOR #32: save map error!!!

mohamedsamirx, 2023-09-14:
when running 
roslaunch erasor mapgen.launch
rosbag play /home/mohamed/02_860_to_950_w_interval_2_node.bag

i got his error
terminate called after throwing an instance of 'pcl::IOException'
  what():  : [pcl::PCDWriter::writeASCII] Could not open file for writing!
[kitti_mapgen-1] process has died [pid 91485, exit code -6, cmd /home/mohamed/catkin_ws/devel/lib/erasor/kitti_mapgen __name:=kitti_mapgen __log:=/home/mohamed/.ros/log/e74d25a6-5357-11ee-b334-557d0f89a6e3/kitti_mapgen-1.log].
log file: /home/mohamed/.ros/log/e74d25a6-5357-11ee-b334-557d0f89a6e3/kitti_mapgen-1*.log

how can be solved ??
thanks

niuwengang, 2023-11-06:
modify
```
 <rosparam param="/map/save_path">"$YOUR_FILE_PATH"</rosparam>
```
 in mapgen.launch

nit-xhg, 2024-05-04:
[image]

nit-xhg, 2024-05-04:
Why is this still the case after modification？
[image]
[image]

closed 2025-04-12 by mohamedsamirx